petersenpeter/CellExplorer #2: Problem launching CellExplorer

nnitzan, 2020-02-06:
Hi Peter, 
I tried to use the new version of CellExplorer, loading the data and metadata and calculating all metrics including mono_res works fine. However, when launching the CellExplorer viewer I get the following error 
"Identifier 'fig' is not a function or a shared variable. To share 'fig' with nested function, initialize it in the current scope. For more information, see Sharing Variables Between Parent and Nested Functions." 
Could it have something to do with the MATLAB version? I'm using 2019b. 

thanks!

petersenpeter, 2020-02-06:
Hi Noam
Did you download the most recent version? Does this happens in the pipeline or separately in the Cell Explorer? It could be a Matlab version issue, that I have not experienced, as the never versions of Matlab are more strict with shared variables. I will see if I can reproduce the problem and update the repository.

nnitzan, 2020-02-06:
it only happens in the CellExplorer, the pipeline runs fine (it creates a session, cell_metrics and mono_res cell infos ). I'm using the most recent version.

thanks!

petersenpeter, 2020-02-06:
I did a quick fix that should solve it.

petersenpeter, 2020-02-06:
Let me know how it goes

nnitzan, 2020-02-06:
wow that was fast, works now! 
thanks!

petersenpeter, 2020-02-06:
Great. Figure handles in Matlab can be a headache to handle... :-)

closed 2020-02-06 by petersenpeter