petersenpeter/CellExplorer #3: monosynaptic connections wrongly identified

nnitzan, 2020-03-10:
Hi Peter, 

For some reason the MonoSyn routine doesn't identify inhibitory cells, even in cases where the inhibition is very clear (see attached image) 
On the other hand, the algorithm occasionally identifies cells that are classified as interneurons as 'excitatory', which could be due to some rebound effects (see second image). 
Any idea how to fix these issues? should these CCGs be deselected? 

Another issue is, when I try to save a classification as a new file I get the following error: 

Reference to non-existent field 'spikeSortingID'.

Error in CellExplorer/saveDialog (line 6819)
                if length(unique(cell_metrics.spikeSortingID)) > 1
 
Error using uiwait (line 81)
Error while evaluating Menu Callback.

at what point is this field  'spikeSortingID' generated? does it depend on what program I used for the sorting? (it happens both with sessions that were manually curated with phy and klusters) 

thanks in advance, 

Noam
[image]
[image]

petersenpeter, 2020-03-10:
Hi Noam. The inhibitory connections are not saved in the current state. I believe it was a decision by Sam to take that out. I will be happy to implement it back into the pipeline. 

You have to manually deselect the false detected connections. A better solution is to increase the significance level, which I believe is set too low right now. 

Regarding your saving error: replace the if line with below
REPLACE THE LINE:
if length(unique(cell_metrics.spikeSortingID)) > 1
WITH:
if UI.BatchMode

I will update the main branch later today with a fix, plus a bunch of new plotting options.

I will let you know once I have a solution for inhibitory connections.

Do the title and buttons always get misplaced for you in the MonoSyn interface as shown in your screenshot? I have not seen this happen in windows or mac OSX...

nnitzan, 2020-03-10:
Hi Peter, 

ok, thanks. 
The graphics and buttons are indeed a little distorted, also in the CellExplorer GUI. 
I'm using remote desktop (Xfce session) on a linux machine. 

-Noam

petersenpeter, 2020-03-10:
Hi Noam. I updated the main branch. 

I have not added inhibitory connections nor your UI displacements. Could you send me a screenshot of the Cell Explorer?

nnitzan, 2020-03-11:
Hi Peter, 

I realized that the distorted graphics were because of the remote desktop I was using which doesn't fully support OpenGL. 
using a VNC client it looks normal (see attached images) 

thanks,

Noam 
[image]
[image]

petersenpeter, 2020-03-11:
OK. there is still some misplacement of some of the panels, but it looks much better now.

petersenpeter, 2020-03-15:
Hi Noam
I updated the Cell Explorer pipeline such that it will find inhibitory connections. The detection threshold is too low at this point and does require extensive manual curation (it is the same algorithm used for excitatory connections). I also updated and tuned **gui_monosyn.m** and the Cell Explorer to be able to handle inhibitory connections. If you experience issues, please let me know. 

Inhibitory connections are not detected by default by the pipeline (because they are unreliable and rare in CA1), so to do so, change the **calc_CellMetrics** input parameter **includeInhibitoryConnections** to true. 

If you run the pipeline on a session with an existing mono_res file that file will not be overwritten. So to test this on an existing session, rename the mono_res file and rerun the algorithm to have it detect the inhibitory connections.

E.g. from the basepath of a session type:
calc_CellMetrics('includeInhibitoryConnections',true)

Let me know how it goes. 

- Peter

petersenpeter, 2020-03-15:
Btw, the monosynaptic algorithm does falsely detects rebound inhibition as a direct inhibitory connection, I will see if I can find a solution for this.

petersenpeter, 2020-03-16:
Hi  Noam
I updated the repository with some minor improvements to the monoSyn interface. You can now distinguish direct inhibitory connections from rebound inhibition and more easily determine the direction of the connection.

[image]

nnitzan, 2020-03-18:
Hi Peter,

thanks, I'll test it in the next few days and give you an update.

-Noam

[monosyn]<https://user-images.githubusercontent.com/14089401/76806905-0c3a9d00-67b9-11ea-9d04-80f1d4604e82.png>

nnitzan, 2020-03-23:
Hi Peter, 

I tested the inhibitory connection detection and it's works pretty well. As you said, it requires a lot of manual curation, though. 
One problem I noticed it that if you want to manually adjust an inhibitory connection via the CellExplorer GUI, once selecting 'inhibitory connections' in the top, it jumps to a different reference cell than the one selected. 

btw - having the channel map next to the waveforms is super helpful! 

Thanks,

Noam 
[image]

petersenpeter, 2020-03-23:
Thanks Noam. I will change the behavior when switching between E and I connections. 

The high false-positive rate is due to the statistical test is not optimal for I connections.  I can decrease the significance level for the inhibitory connections, but the current algorithm does not handle strong E connections very well when detecting I connections.

petersenpeter, 2020-03-23:
Btw, there are a lot of new interaction modes. Press H in the monoSyn GUI to learn more.

petersenpeter, 2020-03-23:
I updated the repo with a fix for the switching between E and I connections.

petersenpeter, 2020-03-26:
Noam. I will close the issue, you are welcome to reopen it if you are still experiencing issues.

closed 2020-03-26 by petersenpeter