petersenpeter/CellExplorer #4: Waveforms from bad channels not excluded

nnitzan, 2020-04-16:
Hi Peter, 

I have a bad channel on the probe and labeled it in the session metadata as 'Bad'. 
However, this channel is still used for the waveform extraction and because the large noise amplitude, is selected as the channel with largest waveform amplitude for display in CellExplorer. 
Are there any extra steps to do in order to avoid this during the creation of the sessionInfo or is it a bug? 

best,

Noam 
[image]
[image]

petersenpeter, 2020-04-16:
Hi Noam. 

Thanks for reaching out with your issue(s). It seems it was a bug, as the session struct was not provided for the waveform extraction. I just updated the repo and the issue should be fixed. 

You can delete your .spikes file and forceReload the waveforms in ProcessCellMetrics (or just delete the cell_metrics file as well, and run the ProcessCellMetrics again).

What are your monitor's resolution and size? I keep seeing that the UI is not behaving as it should on your screenshots.

- Peter

nnitzan, 2020-04-16:
Hi Peter,

Thanks the issue seems to be fixed now, as judged by the summary figures, but after updating the repository I get the following error message

Reference to non-existent field 'plotInsetChannelMap'.

Error in CellExplorer (line 410)
if UI.settings.plotInsetChannelMap; UI.menu.display.showChannelMap.ops(UI.settings.plotInsetChannelMap).Checked = 'on'; end

Error in ProcessCellMetrics (line 1316)
CellExplorer('metrics',cell_metrics,'summaryFigures',true,'plotCellIDs',-1); % Creates group plots only

-Noam

petersenpeter, 2020-04-16:
I forgot to push the preference file to Github. Could you update your repo and try again?

nnitzan, 2020-04-17:
Thanks! it's working well now

petersenpeter, 2020-04-17:
Great. I forgot that I changed filenames and some of the folder structure so I hope it did not cause any further problems for you. The UI should be more responsive after your update.

closed 2020-04-17 by petersenpeter

nnitzan, 2020-04-17:
Thanks Peter,

Fyi - I noticed another minor bug in ProcessCellMetrics (line 1316) when plotting the connectivity graph in the summary plots if there aren't any connections detected during in the MonoSyn the function crashes and cell_metrics is not extracted

best,

Noam

Noam Nitzan
Schmitz Lab
Charité University Medicine Berlin
Charité Crossover NWFZ, Campus Mitte
Virchowweg 6 - Charitéplatz 1
10117 Berlin, Germany

Phone: +49-30-450-639-063
http://schmitz.neurocure.de

petersenpeter, 2020-04-17:
Thanks Noam. I will fix it today.