petersenpeter/CellExplorer #5: what is the required structure of xml files?

saman-abbaspoor, 2020-05-17:
Hello,
I am trying to use `CellExplorer` for my own dataset. Although on the website, it is mentioned that XML files are not required, Matlab gives me an error when I do not include the XML files in the clustering path.

`Undefined function or variable 'xml';
Error in loadSpikes (line 201)
                spikes = GetAllWaveformsFromDat(spikes,xml,basepath,basename,LSB,session)
`

We are using the `OpenEphys` system for recording which gives an XML file and I included that into the clustering path but then other errors appear.

`Reference to non-existent field 'AnatGrps'.
Error in loadSpikes>GetAllWaveformsFromDat (line 569)
if isfield(xml.AnatGrps,'Skip')`

I am using `Klusta` for spike sorting. Is there any suggestion for how this might be fixed?

Also, can you provide details on what the required XML filed should contain?

petersenpeter, 2020-05-19:
Hi Saman,

Sorry for the poor documentation on this aspect of the dependencies. There is a  dependency on the XML file when extracting the waveforms. I will take that dependency out very soon. But in your case, you can easily fix it by adding the missing XML file. I am not sure about what XML file that OpenEphys creates, but the XML file should be compatible with Neuroscope. It must contain the probe layout as defined as AnatGrps and spike groups. Otherwise, you can create the file by downloading and installing NDManager or Neuroscope. http://neurosuite.sourceforge.net/

 If you provide me with more information about your recording I can help create the file for you.

For the spike sorting format please select KlustaViewa (klusta).

petersenpeter, 2020-05-27:
Hi Saman

I updated the repository such that it does not rely on the .xml file anymore. You still have to provide a sufficient level of metadata through the session struct. But please see this tutorial: https://cellexplorer.org/tutorials/metadata-tutorial/

I will be happy to talk by email or try a web call if that would be of help.

saman-abbaspoor, 2020-05-28:
Dear Peter,

Thank you so much for your update. I will try it on my dataset and will get back to you. I managed to create the .XML files but still had issues with running the toolbox. I will see if the new changes will solve that.

Thank you again for your time.

closed 2020-06-20 by petersenpeter