petersenpeter/CellExplorer #15: Waveforms in datasets with different acquisition rates

ThomasHainmueller, 2020-10-05:
Hi Peter, there's a minor bug upon trying to load datasets with different sampling rates (e.g. 20 & 30 kHz) into the same session, due to a mismatch in the length of 'waveforms'. Could potentially be resolved by up- or downsampling one of the waveform sets or by storing them in a cell array.

petersenpeter, 2020-10-06:
Hey Thomas!
CellExplorer already determines an optimal (resampled) waveform if cells have different waveform lengths or sampling rates. If this is not working properly let me know the path to the sessions you are using and I can try to debug.

petersenpeter, 2020-10-06:
By the way Thomas. If you fix bugs or make improvements, feel free to make a pull request.

closed 2020-10-25 by petersenpeter