petersenpeter/CellExplorer #17: Is there a way to upload sorted data from ultramegasort2000 mat files into cell explorer?

shirquinn, 2020-10-23:
Is there a way to upload sorted data from ultramegasort2000 mat files into cell explorer?

ultramegasort2000 github
https://github.com/danamics/UMS2K -

example mat files of ultramegasort sorted spikes:
https://drive.google.com/file/d/1h7VBAiu2oFu9pDtBcFjpI8EA9E4uJA1Z/view?usp=sharing 

spikes = 
  struct with fields:
             params: [1×1 struct]
               info: [1×1 struct]
          waveforms: [12907×45 single]
         spiketimes: [1×12907 single]
             trials: [1×12907 single]
    unwrapped_times: [1×12907 single]
            assigns: [1×12907 double]
             labels: [8×2 double]

petersenpeter, 2020-10-24:
Hi 

I just added support for ultramegasort2000 as per your request using the dataset, you linked to, as a template. Let me know how it works for you. 

The support is handled in the script loadSpikes. To specify the ultramegasort2000 data format, you must select it as part of the session struct for the session or directly when creating the spikes struct with the loadSpikes script. e.g.: go to the data folder in MATLAB and type:
loadSpikes('basepath',pwd,'clusteringformat','ultramegasort2000','getWaveformsFromDat',false)
% getWaveformsFromDat is set to false, as the dataset did not contain any raw data. If you have the raw data you can leave that out and the script will pull out the waveforms from the trace.

You can also create the struct when you run ProcessCellMetrics using the GUI:
ProcessCellMetrics
%Make sure to select the correct sampling rate and channel configuration and deselect getWaveformsFromDat on the first tab (shown in the screenshot below):

[image]

petersenpeter, 2020-11-04:
Hi. I will close this issue now, but you are welcome to open it again if you experience any issues.

closed 2020-11-04 by petersenpeter

shirquinn, 2020-11-04:
Hi, thank you, i think it works fine
Shir Quinn
MD-Phd student
Rubinstein lab
Goldschleger Eye Research Institute
The Department of Human Molecular Genetics and Biochemistry
Sackler School of Medicine, Tel Aviv University , Israel
Tel: (+972) 054-7839444

shirquinn, 2020-11-04:
Thank you, I think it works fine