petersenpeter/CellExplorer #21: adding waveclus support

shirquinn, 2020-12-14:
Hi
Thank you again for creating this amazing tool.
Is there an option to add support for waveclus aswell?
https://github.com/csn-le/wave_clus

The output of Wave_clus  is **times_[filename].mat** containing the following variables: 
par (parameters used for clustering) 
spikes (a matrix with the spike shapes)
inspk (a matrix with the features of the spike shapes)
cluster_class (a matrix with the clustering results).
The variable cluster_class has 2 columns and nspk rows (nspk is the number of spikes). 
The first column is the cluster class, with integers denoting the clusters membership and a value of 0 for those spikes not assigned to any cluster. 
The second column is the spike times in ms.

output files example: 
https://drive.google.com/file/d/1PCrOGmbOFkEvO5QyqxqrwAI96I6c1Kkc/view?usp=sharing
data files:
https://drive.google.com/file/d/1NrsHZ_MdUYPUgdfTnSLjruAmrp7fSJ9u/view?usp=sharing

either way thank you

petersenpeter, 2020-12-15:
Hi

I will be happy to implement wave_clus in CellExplorer. I cannot access the linked files, but I requested access, so once I have this, it should not take long. 

/ Peter

shirquinn, 2020-12-15:
Thank you! I've changed the sharing settings
the waveclus files:
https://drive.google.com/file/d/1PCrOGmbOFkEvO5QyqxqrwAI96I6c1Kkc/view?usp=sharing

data:
https://drive.google.com/file/d/1NrsHZ_MdUYPUgdfTnSLjruAmrp7fSJ9u/view?usp=sharing

In cluster_class , cluster 0 contains spikes not assigned to any cluster, can you take in cell explorer only clusters with the class value of 1 and higher?

petersenpeter, 2020-12-18:
Hi

I updated the github repo with support for Wave_clus. Could you please check and see if it works as expected? It runs fine on the dataset (wave_clus files) you shared with me. I could not download the raw data unfortunately. 

Run this command from the data path:
`loadSpikes('format','Wave_clus','getWaveformsFromDat',false)`
getWaveformsFromDat is false as I could not download the raw data

Next run:
`ProcessCellMetrics`
This should show the session GUI. Make sure Wave_clus is selected as the spike sorting format.

shirquinn, 2020-12-19:
Works great, thank you!

closed 2020-12-19 by petersenpeter

ferchaure, 2022-01-07:
I tried today without luck, It looks like the wave_clus support it's quite broken

petersenpeter, 2022-01-07:
Fernando, could you provide more details on the issue? Perhaps share your data with me so I can test it? 

- Peter

ferchaure, 2022-01-07:
Hi Peter, loadSpikes doesn't set the struct `session.extracellular` for that sorter. After patching that I tried to use ProcessCellMetrics and got multiple errors, but maybe that just my patch... I tried to bugfix a bitmore but I'm not familiar with this software to do much.

I just shared with you a data example