petersenpeter/CellExplorer #22: get correct waveforms for each unit

rajatsaxena, 2020-12-18:
I am trying to load data collected using UCLA silicon probes (128DN) and sorted using `kilosort2` and `phy`. I followed the guidelines mentioned in the `General tutorials`. But when I run the `cell_metrics = ProcessCellMetrics('session', session);`, I run into issues with the `loadSpikes` and `getWaveformsFromDat` functions. 

The script loads the first few waveforms correctly, and then the following unit waveforms aren't real waveforms and look like noise. It seems like a channel map/ spike group/ or electrode group definition related problem to me. Can you please point me to the instruction to define either of those things? Or let me know if my question is unclear. Thank you!

petersenpeter, 2020-12-18:
Hi Rajat

`getWaveformsFromDat` determines the best channel(s) to read out average spike waveforms. It takes 600 spikes from each unit and determines the peak channel. This process is sensitive to bad channels (dead/floating/high impedance) as noise fluctuations can be quite high in amplitude. You, therefore, must define the [bad channels](https://cellexplorer.org/datastructure/data-structure-and-format/#session-metadata) beforehand, e.g.: `session.channelTags.Bad.channels = 1:5;`

From your second screenshot, it is difficult to see if this is really the issue though, as I don't see any real waveforms in the top left plot with "Filtered waveforms across channels" and the noise waveforms it picks up are quite low in amplitude. But other than this the `getWaveformsFromDat` script should work. Could it be that the spike timestamps are not corresponding to the local timestamps of the data file? What hardware did you use to collect the data and what is the sampling rate of the system?

I will be happy to run your data through the processing pipeline if you can share the session with me.

rajatsaxena, 2020-12-18:
I used Intan RHD Recording System to acquire the data with a sampling rate of 30KHz. The data was sorted using kilosort2.5 and phy. I have uploaded it here: https://drive.google.com/drive/folders/1q4w2rgtf5W_0FIdy8sIDu2cV1faPNNuD?usp=sharing

Let me know if you need more information from here.

petersenpeter, 2020-12-18:
Thanks. I will take a look at the data

petersenpeter, 2020-12-19:
I could process your data, though I made some adjustments to the script `sessionTemplate.m`, such that it now pulls out metadata from KiloSort's rez.mat file (your dataset had a rez2.mat file). I believe your problem could have been incorrect metadata.

Pull the latest version of CellExplorer from GitHub and try the same session again and let me know how it goes. Make sure to delete any files generated from CellExplorer with the dataset beforehand.

rajatsaxena, 2020-12-19:
I figured I messed up the metadata, but it works perfectly now. Thank you so much for such a fast response and fix.

closed 2020-12-19 by rajatsaxena