petersenpeter/CellExplorer #23: Problem accessing DataBank

jkopsick, 2020-12-21:
Hi Peter,

I am a PhD student at George Mason University utilizing the CellExplorer MATLAB GUI, and have ran into an issue with loading a session via the function db_load_sessions (using a Windows 10 PC). First off, I am able to successfully load the example session as delineated [here](https://cellexplorer.org/publicdata/preparation/). However, I receive an error when setting the session when following step 4 of the [database tutorial](https://cellexplorer.org/tutorials/database-tutorial/). Following the database tutorial, for public access we can ignore steps 1-3 -- if I do so, then I receive an error that the repository in db_local_repositories is empty, which makes sense. However, when I set it to the repository path for Windows in db_local_repositories, I get the error of "db_set_path: Unable to change directory to basepath: Z:\Buzsakilabspace\Datasets\PetersenP\MS13\Peter_MS13_171129_105507_concat". 

Can you point me in the right direction as to solving this error? I'm thinking I may have missed something obvious, or perhaps the path to the database changed? I'm also happy to send screenshots if necessary.

Thank you for your time and for creating this wonderful tool alongside your colleagues in the lab.

Best,

Jeffrey Kopsick

petersenpeter, 2020-12-23:
Hi Jeffrey Kopsick

I will get back to you by tomorrow with a detailed answer.

petersenpeter, 2020-12-28:
Hi Jeffrey

The directions you followed are primarily for Buzsaki lab members at this point as the public database is still under development (and not properly documented yet). The cell metrics for a large set of processed sessions can be downloaded in [CellExplorer in Matlab](https://cellexplorer.org/publicdata/reference-data/). The Matlab tools cannot be used to download the full dataset. You will have to download it via our web [interface](https://buzsakilab.com/wp/database/) and [globus](https://app.globus.org/file-manager?origin_id=558c54fe-cc41-11e8-8c6a-0a1d4c5c824a&origin_path=%2F). Each session has a direct link in the web interface and you can explore all our public data via globus as well.

For the specific error you experience, this is due to that the script tried to change to a local directory with the data that does not exist. If you just want to load the metadata (which you can) you have to provide an extra input: 
```m
session = db_set_session('sessionName', 'Peter_MS13_171129_105507_concat', 'changeDir', false);
```

If you do want to point to a session that you have downloaded, you have to add the path to `NYUshare_Datasets` to `db_local_repositories`, e.g.
```m
repositories.NYUshare_Datasets = '/Volumes/buzsakilab/Buzsakilabspace/Datasets';
```
This will serve as a local repository. Now the `NYUshare_Datasets` has to have the same organization as meaning that the full path to the dataset would be: 
`/Volumes/buzsakilab/Buzsakilabspace/Datasets/PetersenP/MS13/Peter_MS13_171129_105507_concat`
where this corresponds to local_repository/investigator/animal/session

Let me know if you have any further questions of if I should provide further details.

- Peter

jkopsick, 2021-01-13:
Good morning Peter,

After following your instructions with downloading the full dataset, I was able to successfully view a session's metrics. As far as the public database being under development, when might users expect for a first release to be available? 

Thank you very much for your clear instructions, and I will let you know if I have any further questions.

Best,

Jeffrey

petersenpeter, 2021-01-13:
We are working towards putting a lot more of our public data into the database as we are sharing much more data than what is described in the database. You can find all our public data via [globus](https://app.globus.org/file-manager?origin_id=558c54fe-cc41-11e8-8c6a-0a1d4c5c824a&origin_path=%2F) and our [webshare](https://buzsakilab.nyumc.org/datasets/](https://buzsakilab.nyumc.org/datasets/). I  recommend downloading via globus as this will provide must faster transfer and better stability.

What is really missing at this point is the documentation, updating and standardizing the metadata and various file formats, and improving the database web interface. This is all actively in development but will take some time.

I recommend that you tell us if you have an interest in a specific dataset (e.g. from a recent paper or investigator) and we can prioritize this for you.

- Peter

closed 2021-02-20 by petersenpeter