petersenpeter/CellExplorer #24: "Electrodes" field not showing up in gui_session

KaiserArndt2, 2021-02-18:
Hey Peter,

First off I want to say thank you for this awesome package!! It's helped loads!  

Going through the website, your tutorial on the session metadata video had a field for "electrodes" in the extracellular tab.  I am on windows and it doesn't show up for me, and I pulled the most recent updates to the code this morning.

[image]

Not being able to specify electrode and measurements is giving me this error.  

[image]

Any help would be great, and thank you again!

Best,
Kaiser

petersenpeter, 2021-02-20:
Hi Kaiser
The deep superficial calculation needs a probe layout, the error you received should have been a warning and I have pushed a fix for this to GitHub just now. 

The layout is a simple approximation for the vertical location and layout of the sites:
session.analysisTags.probesVerticalSpacing: in units of µm, e.g. 20
session.analysisTags.probesLayout: ['staggered','linear','poly2','poly3','poly5'] 

Check the first lines of the classification_DeepSuperficial script for further help. If you want to use the gui_session to add these, they are added in the analysis tags section (click the tab called tags). Otherwise, the script will use a default layout, but should not break this time.

See this tutorial on the deep superficial classification: 
https://cellexplorer.org/tutorials/deep-superficial-tutorial/

KaiserArndt2, 2021-03-05:
Thank, and sorry for the late reply I was able to get this working.

closed 2021-03-05 by KaiserArndt2