petersenpeter/CellExplorer #36: plot connectivity graph

rajatsaxena, 2021-09-08:
Hi , 

I managed to get the monosynaptic connection between a group of neurons using the following command:
`mono_res = ce_MonoSynConvClick (spikes,'includeInhibitoryConnections',true,'epoch',epochs);`

I was wondering if there is an easy way to just plot the connectivity graph using the `mono_res` structure output. I noticed in the main script (https://github.com/petersenpeter/CellExplorer/blob/master/CellExplorer.m), you make some sort of adjacency matrix (`A`) followed by a function call to `digraph` and finally plotting it in the GUI. 

Thanks

petersenpeter, 2021-09-09:
Hi Rajat

I added an example script to the repository showing how to generate the connectivity graph from the cell metrics. 
The connection pairs must first be translated into a matrix with connections to generate the connectivity graph. 

https://github.com/petersenpeter/CellExplorer/blob/master/tutorials/connectivity_graph_example.m

Let me know if you have problems with running the code. 

- Peter

rajatsaxena, 2021-09-09:
Thanks for the prompt reply. I will give it a shot today and get back to you asap.

closed 2021-09-30 by rajatsaxena