petersenpeter/CellExplorer #42: Loading Phy Data Error

JerryMekdara, 2021-11-09:
Hello @petersenpeter,

Thanks for putting together this awesome CellExplorer package. I just went through all of the tutorials on the site and the software seems pretty straight forward. I tried loading my preliminary data that was sorted with Kilosort and "viewed" in Phy, but the GUI outputs an error when trying to load spikes from rez.mat or from phy (see error below). 

> Error in loadSpikes (line 265)
            disp(['Importing ' num2str(numel(spikes.times)),'/',
            num2str(length(dataArray{1})),' clusters from phy'])

Error in ProcessCellMetrics (line 270)
    spikes{1} =
    loadSpikes('session',session,'labelsToRead',preferences.loadSpikes.labelsToRead,'getWaveformsFromDat',parameters.getWaveformsFromDat,'showWaveforms',parameters.showFigures);

petersenpeter, 2021-11-09:
Hi Jerry, 

This is tricky for me to troubleshoot remotely. But first, it seems like your error log was cut off. Could you provide the full print of the error? 

Second, please send me your basename.session.mat file, containing the metadata, and perhaps share the session you are trying to process, including your raw data file and the phy files/folder. This should allow me to reproduce the error.

Thanks
- Peter

JerryMekdara, 2021-11-09:
Thanks for replying so soon. I shared the folder with you through a box link in your gmail. The error is below:  

Dot indexing is not supported for variables of this type.

Error in loadSpikes (line 265)
            disp(['Importing ' num2str(numel(spikes.times)),'/',
            num2str(length(dataArray{1})),' clusters from phy'])

Error in ProcessCellMetrics (line 270)
    spikes{1} =
    loadSpikes('session',session,'labelsToRead',preferences.loadSpikes.labelsToRead,'getWaveformsFromDat',parameters.getWaveformsFromDat,'showWaveforms',parameters.showFigures);

petersenpeter, 2021-11-09:
It seems the error occurred because your cluster_group.tsv file was empty. Did you not do any manual curation in Phy or perhaps did not save the changes?

I will push an update to the script that will use the labels from cluster_KSLabel.tsv.

petersenpeter, 2021-11-10:
Hi Jerry 

I updated the repo and it should work with your data now. In your session struct, the spike sorting format should be Phy and leave the filename empty, i.e:

`session.spikeSorting{1}.format = 'Phy'`
`session.extracellular.fileName = ''`
 If you specify the filename, make sure to include the extension:
`session.extracellular.fileName = 'sbpro_0__uid1012-10-07-49.dat'`

JerryMekdara, 2021-11-10:
Thanks Peter! 

Yes, I agree that the data needs to be curated at some point, but we are still currently in the testing phase as we are recording from the spinal cord and don't know how the cells are behaving in the context of specific behaviors yet. I'll try it the new update and report back soon.

JerryMekdara, 2021-12-07:
Hello @petersenpeter,

I wanted to come back to this thread to ask you about how CellExplorer processes the *.tsv/*.npy 'cluster' files outputted by Phy. I've been running into a strange issue where Phy shows, for example, 20 single unit spikes after curation but CellExplorer only indicates 6 single unit spikes. Also, how does CellExplorer handle spikes that have mua labels? 

Thanks
/Jerry

petersenpeter, 2021-12-07:
Hi Jerry.

The spike data import is done in the script [loadSpikes (line 176)](https://github.com/petersenpeter/CellExplorer/blob/master/calc_CellMetrics/loadSpikes.m#L176). Only units with the label good is by default imported. You can import other labeled units, by changing the parameter labelsToRead, e.g. 

```m
spikes = loadSpikes('session',session, 'labelsToRead',{'good','mua'});
```

If you want to import all units (or some other extra label), run `loadSpikes` before running the `ProcessingCellMetrics` script. This will create a spikes file in the data folder that ProcessingCellMetrics` will use.

I must add that during the import, there is an exclusion of spikes happening within 0.5 ms that belong to the same unit. This is done to minimize double detected spikes, which is an issue with Kilosort due to noise artifacts.

JerryMekdara, 2021-12-07:
Thank you! I'll try adding the 'mua' labels to loadSpikes.

closed 2022-01-07 by petersenpeter