petersenpeter/CellExplorer #43: Loading CellExplorer with example data

scottkil, 2021-11-16:
I'm really excited to use CellExplorer, but I'm having an issue running it with the example data provided. Below is the following error message I receive when I run CellExplorer('metrics',cell_metrics)

I do have 'CellExplorer\exampleData\cell_metrics_batch.mat' loaded into the workspace, fyi

Thanks

```
>> CellExplorer('metrics',cell_metrics);
Failed to find basepath files
[15-11-2021 23:06:28] Initializing session
[15-11-2021 23:06:28] Initializing dimensionality reduction plot
Error using CellExplorer>@(X)X(1:end-2)
Too many input arguments.

Error in CellExplorer/initializeSession (line 11213)
            customPlotOptions = cellfun(@(X) X(1:end-2),customPlotOptions.m,'UniformOutput', false);

Error in CellExplorer (line 219)
    initializeSession
 >>
```

petersenpeter, 2021-11-16:
Hi 

This could be due to a conflict with another toolbox. 

First, could you provide me with your search path from Matlab? Type `path` in the command window and paste the list of directories below. 

Second, what are your Matlab version and operating system?

scottkil, 2021-11-16:
Thanks for the quick reply Peter. I resolved this issue. I'm not sure what the source of the error was, but I'll briefly describe what I did in case it helps others later. 

I had a previous version of CellExplorer hiding in my MATLAB path somewhere, so I removed that. I also think there may have been an issue when I cloned the repo; it seems like not all of the files copied over. I just downloaded the repo instead and that seems to work just fine now.

I'm running MATLAB 2021a and Windows 10 Pro. Also, the error message in my first comment was the full log. 

Thanks again!

closed 2021-11-16 by scottkil

petersenpeter, 2021-11-16:
Great. Happy to hear.