petersenpeter/CellExplorer #48: FindInIntervals mex not working (linux)

hyr2, 2022-02-20:
**Operating system**
Linux - Fedora

**Matlab version**
2021a

**Describe the bug**
Mex file called FindInIntervals is not working located in CellExplorer/calc_CellMetrics/mex/
Execution of script FindInInterval as a function is not supported:

**Matlab error log**
Execution of script FindInInterval as a function is not supported:

petersenpeter, 2022-02-21:
Hi

Did you compile the c-code functions? 

https://cellexplorer.org/#getting-started

**Getting started**
The pipeline uses two c-code files that must be compiled CCGHeart.c and FindInInterval.c (originally part of the FMA toolbox). Compiled versions are included for Windows 64bit and Intel Mac 64bit, but you still have to compile them if your OS version is different. If you are using Linux you have to compile the scripts. In MATLAB, go to CellExplorer/calc_CellMetrics/mex/ and run these line:
mex -O CCGHeart.c
mex -O FindInInterval.c

hyr2, 2022-02-21:
Thanks. That solved the problem I was having.

closed 2022-02-21 by hyr2