petersenpeter/CellExplorer #51: how to deal with multiple raw files for basename

YangfanPeng, 2022-03-29:
Hello Peter,

I am trying to prepare my data to be read into the cellexplorer data structure. I understand that the raw dat file should also be in the basepath. However, I have multiple Neuropixels probes per session and thus more than one raw dat file. 

1. Do I create 2 folders for 2 sessions, e.g. /yp012_220325_npx1 and /yp012_220325_npx2 that include the different raw files yp012_220325_npx1.dat and yp012_220325_npx2.dat? Or can I have 1 session metadata file that reads in multiple raw data files?
2. As the raw files are quite big, we would like to store them in a separate directory which is also read only. Is there a way to point cellexplorer to a different relative path for the raw data?
3. Another edge case for another person in the lab would be different raw data files within one recording "session" that have different task paramters. How does one deal with this? I have seen that one could add multiple probes etc to one session metadata struct file?

Maybe also, what is the rawdata file used for? Can I leave it out?

Thanks for your help!

Best
Yangfan

petersenpeter, 2022-03-29:
You have to provide one raw data file at this point. CellExplorer expects it to be called `basename.dat`, but you can indeed provide a relative path to it, e.g.:
```m
session.extracellular.fileName = 'KiloSort_output/amplifier.dat';
```
So the full path will become: `basepath/KiloSort_output/amplifier.dat`. It has to be a relative path. You should therefore concatenate your dat files before running CellExplorer. There is a function included for this called `concatDatFiles`.

I know that with the Neuropixels probes, the data demand has increased a lot, so there is room for improvement here. 

The necessary steps could look like: 
```m
sources = {'yp012_220325_npx1/yp012_220325_npx1.dat','yp012_220325_npx2/yp012_220325_npx2.dat'};
target = 'yp012_220325.dat';
concatDatFiles(sources,target);
session.extracellular.fileName = 'yp012_220325.dat';
cell_metrics = ProcessCellMetrics('session',session);
```
You can then delete the concatenated dat file once processed.

YangfanPeng, 2022-03-29:
Hi Peter,
thanks for the quick reply. I have followed your advice and created a concatenated dat file from my 2 raw files.
One side "bug" report: As I had to copy the files from different locations, I needed to but the folder path in "...", as suggested from this page: https://ss64.com/nt/copy.html

> Copy from a different folder/directory:
> COPY "C:\my work\some file.doc" "D:\New docs\newfile.doc"

But now I am kind of stuck conceptionally... How would I run the cell metrics pipeline when I have two folders containing kilosort output from the two different probes and one concatenated raw file?

Also, how would the cellexplorer distinguish between the two different probes and their channels?
Regarding the third question, maybe we can put it on hold, as it might get solved when I am bit clearer on how your pipeline handles different raw data in one session. Thanks for your help!

Best
Yangfan

petersenpeter, 2022-03-29:
You have to explain this in a bit more detail. Are yp012_220325_npx1.dat and yp012_220325_npx2.dat from the same probe but recorded in serial, or are the dat files from two different probes recorded in parallel?

YangfanPeng, 2022-03-29:
Sorry for being unclear. The latter is true: I have 2 Neuropixels probe simultaneously in the mouse and am recording them at the same time in parallel.

petersenpeter, 2022-03-30:
OK, then my other advice is not relevant. If you recorded them in parallel and spike sorted them separately, I would recommend that you also process them separately in CellExplorer. This is what CellExplorer currently supports best (CellExplorer works with a single raw data file). I have worked with multiple Neuropixels probes for the data from the Allen Institute, but the pipeline is not straightforward. See how loadSpikes handles their 6 neuropixels probes recorded in parallel with multiple dat files for each probe: https://github.com/petersenpeter/CellExplorer/blob/master/calc_CellMetrics/loadSpikes.m#L483

Neuropixels recordings are tricky to work with as the raw data files have different sampling rates and temporal offsets between them. 

I would recommend treating each of the subdirectories as separate sessions as you did for the spike sorting with Phy. Create a session struct in the subfolders and run ProcessCellMetrics separately. This way you will get a cell_metrics file for each of the probes that you can combine and explore together in CellExplorer:

```m
basenames = {'yp012_220325_npx1','yp012_220325_npx2'};
basepaths = {'yp012_220325_npx1','yp012_220325_npx1'};
cell_metrics = loadCellMetricsBatch('basepaths',basepaths,'basenames',basenames);
cell_metrics = CellExplorer('metrics',cell_metrics);
```

Any curation performed in CellExplorer is saved to the original cell metrics files if you press save in the File menu in CellExplorer.

YangfanPeng, 2022-04-04:
thanks Peter for the info! I got it to run as you said and imported both sessions. As I want to keep the raw files separate from the processed kilosort output and cellexplorer mat files, I am considering to include a "copy raw file to kilosort output folder" code and then delete the raw copy after the ProcessCellMetrics function. Do you think this is a sensible approach or is there a better way? E.g. directly pointing to the raw file that is in a different parent folder?

I also found some issues with the temporal offsets when importing two neuropixels probes. I am currently syncing the timestamps to global_timestamp using a python function in the open ephys toolbox. Are you aware of any MATLAB function? I am still trying to figure it out and will come back when I run into more issues, but I guess this problem is more related to openephys, than cellexplorer. Thanks so much for this great toolbox!

YangfanPeng, 2022-04-04:
I have found a solution to include raw files in a separate parent folder path. I modified these two chunks of code in the `getWaveformsFromDat` function:

```
if isfield(session.extracellular,'fileName') && ~isempty(session.extracellular.fileName)
    fileNameRaw = session.extracellular.fileName;
    filePathRaw = session.extracellular.filePathRaw; %added
else
    fileNameRaw = [basename '.dat'];
end
```

```
if isfield(session.extracellular,'filePathRaw') && ~isempty(session.extracellular.filePathRaw) %checks if filepathraw is defined
    if ~exist(fullfile(filePathRaw,fileNameRaw),'file')
        error(['Binary file missing: ', fullfile(filePathRaw,fileNameRaw)])
    end
    s = dir(fullfile(filePathRaw,fileNameRaw));

    duration = s.bytes/(2*nChannels*sr);
    rawData = memmapfile(fullfile(filePathRaw,fileNameRaw),'Format',precision,'writable',false);
else %original code block
    if ~exist(fullfile(basepath,fileNameRaw),'file')
        error(['Binary file missing: ', fullfile(basepath,fileNameRaw)])
    end
    s = dir(fullfile(basepath,fileNameRaw));

    duration = s.bytes/(2*nChannels*sr);
    rawData = memmapfile(fullfile(basepath,fileNameRaw),'Format',precision,'writable',false);
    % DATA = rawData.Data;
end
```

And then I defined these two variables before I run the pipeline.
```
session.extracellular.filePathRaw
session.extracellular.fileName
```

Not sure if you find this addition useful or not. It serves my use case very well.
Best
Yangfan

closed 2022-04-04 by YangfanPeng

petersenpeter, 2022-04-04:
Hi Yangfan

Happy to hear you got it working. I like your implementation for handling the raw data file outside the basepath. I don't think I can help with the temporal offset, but remember that the sampling rate is not exactly the same for two probes, and their onset differs, you should be able to read their timestamps using the npy matlab toolbox: https://github.com/kwikteam/npy-matlab