petersenpeter/CellExplorer #52: Some problems with the Generate Common Coordinates

tanselbaran, 2022-04-04:
**Operating system**
Ubuntu 21.10

**Matlab version**
R2019b

**Describe the bug**
I tried using the "Extracellular/Generate common coordinates" function in the Session GUI, after having defined probe implants under the Animal Subject. I noticed two problems in the process:

1) In generateCommonCoordinates.m Line 29, it seems to me that the line should be changed from 
`nChannels = session.animal.probeImplants{i}.nChannels;` to `nChannels = session.animal.probeImplants{i}.nChannel;`, since the name of the corresponding variable in `session.animal.probeImplants` is `nChannel`. 

2) In bregma_to_CCF.m, a function named "rodrigues_rot" is used multiple times for performing Rodrigues Rotation. This function is not available either in cellExplorer repository or MATLAB in general. Is that due to the version of the MATLAB I am using? Or am I missing some packages? After a quick Google search, I found [this](https://www.mathworks.com/matlabcentral/mlc-downloads/downloads/submissions/34426/versions/2/previews/rodrigues_rot.m/index.html) script with the name rodrigues_rot.m. If that is the one you are using, maybe it would be helpful to add it into the CellExplorer repository.

Many thanks for this great resource you have developed! 
Best regards,
Baran

petersenpeter, 2022-04-04:
Hi Baran

1: This was a bug, and should remain as nChannels. I updated the probes' metadata that caused this.
2: rodrigues_rot should have been included in the repository. I just added it. Sorry for this. 

I pushed the changes to GitHub. Could you update your local CellExplorer branch/redownload and try again to see if the problems were fixed? 

/ Peter

closed 2022-04-04 by tanselbaran

petersenpeter, 2022-04-04:
2: Great! 
1: Regarding the custom probes. I will be happy to add other commercial or standardized probe information to this metadata model. Just send me the details. The data comes from the buzsaki lab databank (https://buzsakilab.com/wp/silicon-probes/) and can be updated in gui_session from the top menu. See the screenshot of the private submission page below for relevant fields: 

[image]

tanselbaran, 2022-04-04:
Thanks a lot! I'll keep that in mind. Also another question related to the Common Coordinates: Are there any plans or ongoing work of providing inherent support for plotting the rat brain in 3D as well? I was just wondering since there is quite some work done with rats in Buzsaki Lab, if I remember correctly.

petersenpeter, 2022-04-04:
If a 3D Rat atlas is available, it could be implemented. We can do this for the mouse brain because the Allen Institute has made their Atlas public, so if you know of other resources, let me know, and I will see if I can implement them.