petersenpeter/CellExplorer #53: oscillations in waveform

YangfanPeng, 2022-04-05:
Hello Peter,
Sorry to bother you again. In my latest import from kilosort3 (neurpixels) I have weird oscillations in the waveform, see attached image. Do you know where these could be coming from? The raw data looked fine, is it an issue of kilosort3 or is the getwaveform function of cellexplorer the reason? I would appreciate any advice. Thanks!
Best
Yangfan

[image]

petersenpeter, 2022-04-05:
Hi Yangfan

Could it be that the number of channels is not correct? Alternatively, you have a high level of high-frequency noise in your recording...

YangfanPeng, 2022-04-05:
Hi Peter,
thanks for the quick reply! I think I found the error. Something was off in my kilosort batch script. If I run kilosort via the GUI, the input to cellexplorer looks normal. Thanks for the help!

closed 2022-04-05 by YangfanPeng