petersenpeter/CellExplorer #54: ProcessCellMetrics overwriting 'session' input

KaiserArndt2, 2022-04-05:
**Operating system**
Windows 10

**Matlab version**
2018a

**Describe the bug**
In the "deep-superficial by ripple polarity reversal" chunk of 'ProcessCellMetrics', the session struct is being overwritten.  Specifically, session.extracellular.chanCoords.verticalSpacing = 20 is being changed to []. Even when being manual input into the session before running 'ProcessCellMetrics'. This then causes an error in the 'classification_DeepSuperficial' function.

**Matlab error log**
Error using  * 
Incorrect dimensions for matrix multiplication. Check
that the number of columns in the first matrix matches
the number of rows in the second matrix. To perform
elementwise multiplication, use '.*'.
Error in classification_DeepSuperficial (line 275)
deepSuperficial_ChDistance3 =
deepSuperficial_ChDistance3 * VerticalSpacing;
Error in ProcessCellMetrics (line 848)
            classification_DeepSuperficial(session); 

**Session struct**
[m50_190724_181515_HPC.session.zip](https://github.com/petersenpeter/CellExplorer/files/8421826/m50_190724_181515_HPC.session.zip)

**Screenshots**
N/A

**Additional context**
Add any other context about the problem here.

**Matlab setpath**
A common problem is interferences with other toolboxes. If possible, please provide your search path by typing `path` in Matlab.

petersenpeter, 2022-04-06:
Hi Kaiser

I believe this was a bug in the session GUI, that I was able to fix. Could you update your CellExplorer repo and try again?

KaiserArndt2, 2022-04-06:
Yep, that fixed it. Thanks Peter!

closed 2022-04-06 by petersenpeter