petersenpeter/CellExplorer #56: conflict between Kilosort3 and CellExplorer

DaniMagyar, 2022-04-14:
Hi, 

Unfortunately Kilosort GUI does not work when CellExplorer is added to Matlab path. When I try to run Kilosort3 in Matlab, I get the following error:

```
>> kilosort
Error using uiextras.Panel
Cannot define property 'Enable' in class 'uiextras.Panel' because the
property has already been defined in the superclass
'matlab.ui.container.Panel'.

Error in ksGUI/build (line 112)
            obj.H.logPanel = uiextras.Panel(...

Error in ksGUI (line 47)
            obj.build(parent);

Error in kilosort (line 18)
h = ksGUI(f);
```

```
>> which uiextras.Panel
C:\Users\dmagyar\Documents\Matlab_toolboxes_DM\CellExplorer\toolboxes\GUI Layout Toolbox 2.3.4\layout\+uiextras\Panel.m  % uiextras.Panel constructor
```

I have to remove CellExplorer from the path every time I use Kilosort3. Do you have any suggestions how to solve this problem?

Thank you,
Daniel

petersenpeter, 2022-04-14:
Hi Daniel 

I believe it is because we use different versions of the same GUI toolbox (GUI Layout Toolbox). 

https://se.mathworks.com/matlabcentral/fileexchange/47982-gui-layout-toolbox

The GUI Layout Toolbox version 2.3.4 is included with CellExplorer, but Kilosort might use a different version? Could you check which version you have installed with KiloSort?

- Peter

DaniMagyar, 2022-04-14:
Thanks for the quick response. I have the 2.3.5 version.

petersenpeter, 2022-04-14:
OK, I will update CellExplorer to use the same version 2.3.5 (the most recent version) of the toolbox then. This should solve the issue. Hopefully by tomorrow.

DaniMagyar, 2022-04-14:
I have tried different versions in the meantime. It seems like Kilosort3 GUI does not work with GUI Layout Toolbox v2.3.4. Thank you for your help.

petersenpeter, 2022-04-15:
Hi Daniel, I updated the CellExplorer repo where I removed GUI Layout Toolbox v2.3.4 and included GUI Layout Toolbox v2.3.5. You will have to update your Matlab set path. Let me know if this fixes your issues.

DaniMagyar, 2022-04-15:
Hi Peter, the update solved the problem, both Kilosort and CellExplorer work perfectly. Thanks!

petersenpeter, 2022-04-15:
Great. I will close the issue but feel free to open it if you experience further issues.

closed 2022-04-15 by petersenpeter