petersenpeter/CellExplorer #57: Generalization for all sampling frequencies

hyr2, 2022-04-19:
**Is your feature request related to a problem? Please describe.**
The script CCG.m assumes a default value of 20KHz for the sampling rate. Many labs record at 25KHz or 30KHz. This needs to be generalized. To generalize it, we need to add an extra input parameter for the function that calls CCG.m: ce_MonoSynConvClick.m

**Describe the solution you'd like**
I am attaching the updated ce_MonoSynConvClick.m so that you can make the appropriate change. I have added an extra input parameter for the sampling rate.  
[image]
[image]

petersenpeter, 2022-04-19:
Hi 

Thanks for the suggestion and code. I incorporated the changes across the various CCG calls, as you suggested, but maintained the variable name `sr` for sampling rate. For historical reasons, the CCG function takes `Fs=1/sr` as input. 

Here is the commit:
https://github.com/petersenpeter/CellExplorer/commit/69c4a1eca9aba5ec8dc6eaee6f75fdc96c7d283e

petersenpeter, 2022-04-25:
I am closing this issue now, but feel free to open it if you still experience issues.

closed 2022-04-25 by petersenpeter