petersenpeter/CellExplorer #58: manipulations with different timescale

DaniMagyar, 2022-05-02:
Hi Peter,

I would like to plot multiple manipulations that have different duration. (e.g. 1 ms vs 10 sec)

Timestamps of different manipulations are stored in the `timestamps` fields of `basename.eventName.manipulation.mat` structs. 
I also provide duration in the `duration` fields like it is suggested on the website.

When I run `cell_metrics = ProcessCellMetrics('session', session);` the program ignores the `duration` fields, instead loads a hard coded duration value from `preferences = preferences_ProcessCellMetrics(session);` .

Would it be possible to use those duration values that I provide in the `duration` fields of the `basename.eventName.manipulation.mat` structs? In this case, PSTHs of different manipulations could have different length and bin size. 

Thank you,
Daniel

petersenpeter, 2022-05-03:
Hi Daniel

The stimulation intervals are defined by the timestamps: `Px2 matrix with intervals for the P events in seconds`. 

I know that this is not that consistent, as the timestamps are really intervals in this case, but this is to maintain backward compatibility.

The duration field is an optional field, but really just captures `diff(timestamps)`. 

https://cellexplorer.org/datastructure/data-structure-and-format/#events

The calculated PSTHs have the same timestamps/bin size and number of bins across the stimulation events (see `calc_PSTH.m`). If you only have a few different durations (e.g. 1ms, 10ms and 30ms), you could create PSTHs separately for each of the durations and just give them different names.

DaniMagyar, 2022-05-03:
Hi Peter,

This helped to create the appropriate PSTHs. Previously I used a Px1 matrix in `timestamps`, that was the problem. Determining the offset of each manipulation results in appropriate PSTHs. Thanks!

closed 2022-05-03 by DaniMagyar