petersenpeter/CellExplorer #59: Batch cell_metrics containing putative connection transmission probabilities

KaiserArndt2, 2022-05-11:
**Is your feature request related to a problem? Please describe.**
When running "loadCellMetricsBatch" the output doesn't contain the transmission probabilities from the individual session cell_metrics.

**Describe the solution you'd like**
Within the batch cell_metrics.putativeConnections have matrixes containing the transmission probabilities of the corresponding putative connections.

petersenpeter, 2022-05-12:
Hi Kaiser
I added the transmission probabilities to the batch script as you requested. Please update your local CellExplorer repo and let me know if you experience any issues. 

/ Peter

KaiserArndt2, 2022-05-12:
No issues on my end.  Thanks again Peter!!

closed 2022-05-12 by petersenpeter