petersenpeter/CellExplorer #60: Hi Peter,Error loading waveforms

jesusmartin-cortecero, 2022-05-25:
**Operating system**
Please provide your operating system

**Matlab version**
Please provide your Matlab version

**Describe the bug**
A clear and concise description of what the bug is.

**Matlab error log**
Please paste your error log here

**Session struct**
If relevant, please provide your session Matlab struct (`basename.session.mat`) by zipping it and dragging it into this text field. This will ease the troubleshooting. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**Matlab setpath**
A common problem is interferences with other toolboxes. If possible, please provide your search path by typing `path` in Matlab.

petersenpeter, 2022-05-25:
Hi Jesús

Could you share a data session with me that gives this error? It will make it easier for me to troubleshoot. Thanks.

/ Peter

jesusmartin-cortecero, 2022-05-25:
Hi Peter

here is an old file i am using to test cell explorer and that generates the above mentioned error

https://drive.google.com/drive/folders/1wk5Vk9gmgr_xuDY6J-LdwWxVNCZrfsLn?usp=sharing
 

Jesús

petersenpeter, 2022-05-25:
Hi, I don't see a binary/raw data file in the folder?

jesusmartin-cortecero, 2022-05-25:
the name is a bit long, sorry if it is confusing

"010719_Jesus_Emilio_Jittering_1600_1500_3800_h5_diff_laser_duration.bin"

jesusmartin-cortecero, 2022-05-25:
Thanks Peter, yes now is working. 

I just wonder why it doesnt work with 30030Hz? when I enter 30000Hz in spike 2, the program readjust the sampling frequency automatically for all channels including the triggers of the laser and airpuff.

Regarding the triggers, i would like to know how i can enter the triggers of the laser and the airpuffs in Cell explorer to make custom Raster plots and PSTHs. 

Jesús

closed 2022-05-27 by jesusmartin-cortecero

petersenpeter, 2022-05-27:
I would recommend you to work with the rounded sampling rate of 30,000Hz, and then rescale if needed, but I will look into what caused the crash. Every amplifier has its own sampling rate, which is never really 20KHz or 30Hz (say 29987.44Hz for a Neuropixels probe), so this becomes tricky to handle when working with data from multiple recordings.

jesusmartin-cortecero, 2022-05-27:
Thank you very much. 
Jesus

petersenpeter, 2022-05-27:
I fixed the issue, which was a simple plotting error, and committed the changes to GitHub.

CellExplorer generates the PSTHs using:
```m
PSTH = calc_PSTH(event,spikes)
```
The event input follows the event structs convention: https://cellexplorer.org/datastructure/data-structure-and-format/#events

If you want CellExplorer to generate it for you, you simply have to save the trigger events as event files in the basepath before running ProcessCellMetrics. Check the above link and the example dataset here: https://cellexplorer.org/datastructure/data-structure-and-format/#example-dataset

jesusmartin-cortecero, 2022-05-31:
Thank you very much, now is working perfect