petersenpeter/CellExplorer #88: Large Refractory Period Violation Values

saman-abbaspoor, 2022-12-28:
Hi Peter,

I hope you're doing well. First let me thank you again for the great tool you created. I installed the latest version of Cell Explorer and now when I look at the refractoryPeriodViolation, the values are way off. Then I checked the code and I saw that in the formulation, the value is multiplied by 1000. Shouldn't it be 100 if it's meant to convert it to percent? Or maybe I'm missing something? When I calculated the RPVs myself (total number of spike intervals < 2ms divided by total number of spikes), I get the same value but of an order of magnitude smaller than what cell explorer gives me. I checked the spike times from cell explorer and they are correct.

`cell_metrics.refractoryPeriodViolation(j) = 1000*length(find(diff(spikes{spkExclu}.times{j})<0.002))/spikes{spkExclu}.total(j);
`

Windows 10
Matlab R2018b

petersenpeter, 2022-12-28:
Hi Saman

The refractory period is defined per thousand (‰). I must admit that I don't recall why it was defined this way, but it likely follows the literature.
https://cellexplorer.org/datastructure/standard-cell-metrics/#sorting-quality-metrics

Are you also saying that the calculations are not correct or is it only the definition you were uncertain about?

saman-abbaspoor, 2022-12-29:
Thank you Peter,

I wasn't sure about the definition. I thought it's based on percent so the calculations seemed off but with this explanation, it now makes sense. Now I have to search the reason why it's defined per thousand (‰).

closed 2023-01-10 by petersenpeter