petersenpeter/CellExplorer #89: Fitting equation in fit_ACG.m?

PierreLeMerre, 2023-01-10:
Why the fitting equation in fit_ACG.m is slightly different from the one in the website (parenthesis after the first parameter c (decay amplitude))? 
fit_ACG.m -> `max(c*(exp(-(x-f)/a)-d*exp(-(x-f)/b))+h*exp(-(x-f)/g)+e,0)`
website -> `max(c*exp(-(x-f)/a)-d*exp(-(x-f)/b)+h*exp(-(x-f)/g)+e,0)`

petersenpeter, 2023-01-10:
Hi Pierre

Thanks for pointing out this issue! I believe it is a typo that I introduced into the code about four years ago. The equation shown on the website is therefore the "correct equation".

Unfortunately, it is a typo that has now existed in the code for many years now, so to maintain backward compatibility with previously processed data, I will keep the definition from the code and have updated the shown equation on the website. The implication is not so severe, as it only affects the calculation of the parameter d (the amplitude of the rise exponential, which is scaled relative to c). 

Will this work for you?

PierreLeMerre, 2023-01-10:
Hi Peter,

Thanks for the quick reply. I see. I stumbled upon this while I was trying to port the ACG fitting in Julia. Nothing major for sure but somehow the fitting is better with the typo version than the correct one (I just had a quick try on Matlab)! I don't know the reason (as I don't properly understand the algorithm beyond the fitting procedure, difference between 'Levenberg-Marquardt' or 'Trust-Region'). 

Good to know that is a known bug. I'll carry on with the typo version of the fitting and scale down the d parameter relative to c afterwards.

Best.

closed 2023-01-10 by PierreLeMerre

petersenpeter, 2023-01-10:
OK great. Unfortunately, I don't recall if it was an active decision on my part, but it definitely did not reflect what was documented on the website. I did go through the Matlab code just now and the equation was consistently written/defined like this. 

Regarding the performance of the fit, it might simply be how you define the limits of the parameters. 

Best