petersenpeter/CellExplorer #100: Error when saving cell metrics as .nwb file through GUI

slcalgin, 2023-10-31:
**Operating system**
Windows 10 Pro

**Matlab version**
R2023a

**Describe the bug**
There is an error when trying to save cell metrics as an .nwb file through the GUI. 

**Matlab error log**
> In saveCellMetrics2nwb (line 18)
In saveCellMetrics (line 23)
In CellExplorer/saveMetrics (line 9961)
In CellExplorer/saveAsDialog (line 9930)
In uiwait (line 81)
In CellExplorer (line 392)
In main_CellExplorer (line 18) 
Unrecognized function or variable 'generateExtension'.

Error in saveCellMetrics2nwb (line 18)
generateExtension('ndx-cell-explorer.namespace.yaml');

Error in saveCellMetrics (line 23)
            saveCellMetrics2nwb(cell_metrics,file);

Error in CellExplorer/saveMetrics (line 9961)
            saveCellMetrics(cell_metrics,file);

Error in CellExplorer/saveAsDialog (line 9930)
            saveMetrics(cell_metrics,fullfile(SavePath,file));
 
Error while evaluating Menu Callback.

**Screenshots**
[image]
**Additional context**
Add any other context about the problem here.

petersenpeter, 2023-11-05:
Hi

Make sure to install the NWB Matlab toolset beforehand: https://github.com/NeurodataWithoutBorders/matnwb

Follow this tutorial to work with the nwb format: https://cellexplorer.org/tutorials/nwb-tutorial/

Let me know if this fixes your issue.

slcalgin, 2023-11-07:
thank you, I've added the toolset. 
I intend to save a batch cell_metrics, and it now sends this error, that makes sense, the batch cell_metrics.general doesn't have the "basepath" field but basepaths... 
>> cell_metrics.general.basename
ans =
    'batch of sessions'

>> nwb_file = [cell_metrics.general.basename, '.cellmetrics.cellinfo.nwb'];
>> nwb = saveCellMetrics2nwb(cell_metrics, nwb_file);
Unrecognized field name "basepath".

Error in saveCellMetrics2nwb (line 22)
    'session_description', cell_metrics.general.basepath, ...

should I save the cell_metrics for each session to nwb, one at a time?

slcalgin, 2023-12-05:
Hi Peter,

No worries! I've sent you an email with my dataset. I'll try saving my sessions individually in the meantime. 

Thanks!

petersenpeter, 2024-02-28:
Hi

I closed the issue. Feel free to open it again.

closed 2024-02-28 by petersenpeter