petersenpeter/CellExplorer #102: Request for CCGHeart.mexmaca64

RunjieTomYi, 2023-12-08:
Hi Peter,
I was trying to compile mex on my mac and XCode 15(the only version MacOS14 can open) is not supported by MATLAB. So running the mex line generates following error. 
I'm wondering if it's possible for you to compile a CCGHeart with mexmaca64 format. Any other possible solutions are appreciated.

**Operating system**
MacOS Sonoma 14.0

**Matlab version**
R2023b

**Matlab error log**
Building with 'Xcode with Clang'.
Error using mex
/Users/tomyi/Research/CellExplorer-master/calc_CellMetrics/mex/CCGHeart.c:195:3: error: call to undeclared library function
'memcpy' with type 'void *(void *, const void *, unsigned long)'; ISO C99 and later do not support implicit function
declarations [-Wimplicit-function-declaration]
                memcpy(mxGetPr(plhs[1]), (void *)Pairs, PairCnt*sizeof(unsigned int));
                ^
/Users/tomyi/Research/CellExplorer-master/calc_CellMetrics/mex/CCGHeart.c:195:3: note: include the header <string.h> or
explicitly provide a declaration for 'memcpy'
1 error generated.
 
**Screenshots**
[image]

petersenpeter, 2023-12-12:
Hi 

I updated the repository with a compiled version for Apple Silicon (CCGHeart.mexmaca64 and FindInInterval.mexmaca64). Could you test and see if this works for you?

RunjieTomYi, 2023-12-12:
It's working fine now. Much thanks!

closed 2023-12-12 by RunjieTomYi