petersenpeter/CellExplorer #106: Plotting averages of events

dlinsen1, 2024-04-23:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I cannot evaluate the average voltage trace corresponding to events as i am evaluating them for accuracy in NS2 after detection.

**Describe the solution you'd like**
I'd love a plot inset into the GUI showing the mean +/- SEM and/or STD of all of the events. Bonus would be a customizable time window range around said events and an easy way to enter or omit certain events like is done with the channel tags. Essentially a PSTH with some stats baked in.

**Describe alternatives you've considered**
extracting events in matlab and plotting manually - not ideal to see in real time how each event impacts average.

**Additional context**
Add any other context or screenshots about the feature request here.

petersenpeter, 2024-04-23:
Hi 

This is a great idea. But is this a feature request for Neuroscope2 or CellExplorer? 

In NeuroScope2 you can create custom plots: https://cellexplorer.org/interface/neuroscope2-analysis/

dlinsen1, 2024-04-23:
The request is for Neuroscope2. We run a variety of detectors, including bz_findripples, and would like a way to visualize the average of all said events in Neuroscope2 (or certain groups of events) as we go through manually to verify false positives etc. 

Thank you for pointing me to the wrapper section. I am not experienced with using these but will see what I can do.

petersenpeter, 2024-04-23:
Again I think this is a great idea. The wrappers are really like any other function, just make sure to follow the format. 

I will try to find time to do this soonish. :-)