petersenpeter/CellExplorer #108: Issue in windows system

TsungChihTsai, 2024-05-15:
**Operating system**
Windows 11 Enterprise
64-bit operating system, x64-based processor

**Matlab version**
R2023b Update 4 (23.2.0.2428915)

**Describe the bug**
I just run "session = sessionTemplate(pwd,'showGUI',true);", but it showed error. 

I would like to clarify if CellExplore is exclusively compatible with iOS. Because I only saw in the iOS platform in website of CellExplore and videos of YouTube.

Upon visiting the website "https://cellexplorer.org/#getting-started", I noticed indications that it might also be compatible with Windows 64-bit systems. 

**Matlab error log**

**Session struct*
[image]
*

If relevant, please provide your session Matlab struct (`basename.session.mat`) by zipping it and dragging it into this text field. This will ease the troubleshooting. 

I haven't use any data

**Matlab setpath**

	MATLABPATH

	C:\Users\ttsai\Documents\MATLAB
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addon_enable_disable_management\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addon_updates\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons\cef
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons\fileexchange
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons\supportpackages
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_common\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_desktop_registration
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_install_location\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_product
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_product_support\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_registry\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addons_sidepanel\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addressbar_plugins\browse_for_folder_button\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\addressbar_plugins\cd_up_one_dir_button\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appcontainer
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\appdesigner
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\appdesigner\interface
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\appdesigner\runtime
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\appdesigner_api
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\comparisons\mldesktop_m
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\appdesigner\matlab_integration\cfb\cfb
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\apps
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\audiovideo
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\automation\core
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\bigdata
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\bluetooth
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\buildtool\core
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\buildtool\ext
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\capabilities
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\cefclient
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codeanalysis\analyzerrpt\backend
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codeanalysis\reports\analysis
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codeanalysis\tools
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codetools
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codetools\embeddedoutputs
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codetools\embeddedoutputs\DataToolsRegistration
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\codetools\liveapps
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\configtools
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\connector2\binary
	C:\Program Files\MATLAB\R2023b\toolbox\matlab\connector2\common
[... 616 lines cut ...]
	C:\Program Files\MATLAB\R2023b\toolbox\target\shared\devicedrivers_coder
	C:\Program Files\MATLAB\R2023b\toolbox\target\shared\embedded_utilities\common
	C:\Program Files\MATLAB\R2023b\toolbox\target\shared\ioserver
	C:\Program Files\MATLAB\R2023b\toolbox\target\shared\svd\common
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\data\dmsi\matlabonly
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\model\foundation\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\model\target\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\plugins\legacyconnection\matlabprovider\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\user\commandline\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\targetframework\utilities\supportpackage
	C:\Program Files\MATLAB\R2023b\toolbox\textanalytics\BERT\matlab
	C:\Program Files\MATLAB\R2023b\toolbox\textanalytics\livetasks
	C:\Program Files\MATLAB\R2023b\toolbox\textanalytics\textanalytics
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\apps\imageanalyzer
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\apps\mra\src
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\apps\timefrequencyanalyzer
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\apps\waveletsignaldenoiser\src
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\apps\wv1Danalyzer\src
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\compression
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\deep
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\wavelet
	C:\Program Files\MATLAB\R2023b\toolbox\wavelet\wmultisig1d
	C:\Program Files\MATLAB\R2023b\ui\desktop-default-plugins\matlab
>>

TsungChihTsai, 2024-05-15:
I seted path "C:\matlab\CellExplorer-main\CellExplorer-main"
[image]

However, it still did not work. 
[image]

petersenpeter, 2024-05-15:
You need to include all subfolders of CellExplorer to Matlab (except documentation) 

UIX is a graphical toolbox that is included with CellExplorer

TsungChihTsai, 2024-05-15:
I used this add all subfolders 

[image]

And, it works. Thanks!

[image]

closed 2024-05-15 by petersenpeter