petersenpeter/CellExplorer #112: db_is_active not found

evilrobotxoxo, 2024-07-18:
**Operating system**
MacOS Sonoma 14.5

**Matlab version**
R2024a

**Describe the bug**
When I try to load a .dat file, I get an error saying: 

Unrecognized function or variable 'db_is_active'.
Error in => NeuroScope2.m at line 96

**Matlab setpath**
A common problem is interferences with other toolboxes. If possible, please provide your search path by typing `path` in Matlab.

/Users/lukesjulson/Documents/MATLAB
	/Users/lukesjulson/Desktop/github/buzcode
	/Users/lukesjulson/Desktop/github/buzcode/GUITools
	/Users/lukesjulson/Desktop/github/buzcode/GUITools/EventExplorer
	/Users/lukesjulson/Desktop/github/buzcode/GUITools/EventExplorer/EEhelpers
	/Users/lukesjulson/Desktop/github/buzcode/GUITools/TheStateEditor
	/Users/lukesjulson/Desktop/github/buzcode/analysis
	/Users/lukesjulson/Desktop/github/buzcode/analysis/CrossFrequencyCoupling
	/Users/lukesjulson/Desktop/github/buzcode/analysis/RankOrder
	/Users/lukesjulson/Desktop/github/buzcode/analysis/SharpWaveRipples
	/Users/lukesjulson/Desktop/github/buzcode/analysis/SharpWaveRipples/detect_swr
	/Users/lukesjulson/Desktop/github/buzcode/analysis/SpectralAnalyses
	/Users/lukesjulson/Desktop/github/buzcode/analysis/assemblies
	/Users/lukesjulson/Desktop/github/buzcode/analysis/cellTypeClassification
	/Users/lukesjulson/Desktop/github/buzcode/analysis/cellTypeClassification/BrendonClassificationFromStark2013
	/Users/lukesjulson/Desktop/github/buzcode/analysis/lfp_general
	/Users/lukesjulson/Desktop/github/buzcode/analysis/monosynapticPairs
	/Users/lukesjulson/Desktop/github/buzcode/analysis/monosynapticPairs/utils
	/Users/lukesjulson/Desktop/github/buzcode/analysis/placeFields
	/Users/lukesjulson/Desktop/github/buzcode/analysis/positionDecoding
	/Users/lukesjulson/Desktop/github/buzcode/analysis/spikeLFPcoupling
	/Users/lukesjulson/Desktop/github/buzcode/analysis/spikes
	/Users/lukesjulson/Desktop/github/buzcode/analysis/spikes/correlation
	/Users/lukesjulson/Desktop/github/buzcode/analysis/spikes_general
	/Users/lukesjulson/Desktop/github/buzcode/analysis/spikes_general/InterSpikeIntervals
	/Users/lukesjulson/Desktop/github/buzcode/database
	/Users/lukesjulson/Desktop/github/buzcode/detectors
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectBehavior
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectBehavior/detect_pupilDilation
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectBehavior/detect_vocalizations
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectBehavior/detect_whisking
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectEvents
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectStates
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectStates/SleepScoreMaster
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectStates/SleepScoreMaster/ScoringParameters
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectStates/SleepScoreMaster/StateScoreFigures
	/Users/lukesjulson/Desktop/github/buzcode/detectors/detectStates/SleepScoreMaster/TestsAndExamples
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/CONVNFFT
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/CircularStats
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/CircularStats/examples
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/ExportFig
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/Analyses
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/Data
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/Database
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/General
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/Helpers
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/IO
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FMAToolbox/Plot
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/FilterM
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/IoSR-Surrey-MatlabToolbox-4bff1bb
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/IoSR-Surrey-MatlabToolbox-4bff1bb/deps
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/IoSR-Surrey-MatlabToolbox-4bff1bb/deps/SOFA_API
	/Users/lukesjulson/Desktop/github/buzcode/externalPackages/IoSR-Surrey-MatlabToolbox-4bff1bb/deps/SOFA_API/conventions
[... 133 lines cut ...]
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/intanToolbox
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/metadata
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/metadata/BWMetadataSystem
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/old
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/positionTracking
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/positionTracking/LEDTracking
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/positionTracking/Spotter
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/positionTracking/optitrack
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/probeGeometries
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/ratemapping
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/sessionsManagement
	/Users/lukesjulson/Desktop/github/buzcode/preprocessing/signalAlignment
	/Users/lukesjulson/Desktop/github/buzcode/tutorials
	/Users/lukesjulson/Desktop/github/buzcode/tutorials/exampleDataStructs
	/Users/lukesjulson/Desktop/github/buzcode/tutorials/exampleDataStructs/20170505_396um_0um_merge
	/Users/lukesjulson/Desktop/github/buzcode/utilities
	/Users/lukesjulson/Desktop/github/buzcode/utilities/fileConversions
	/Users/lukesjulson/Desktop/github/buzcode/utilities/fileConversions/ConvertToPlexon
	/Users/lukesjulson/Desktop/github/buzcode/utilities/fileConversions/LinuxBasedPCAForFets
	/Users/lukesjulson/Desktop/github/buzcode/visualization
	/Users/lukesjulson/Desktop/github/buzcode/visualization/colormaps
	/Users/lukesjulson/Desktop/github/buzcode/visualization/colormaps/ColorPaletteGenerators
	/Users/lukesjulson/Desktop/github/buzcode/visualization/colormaps/crameri
	/Users/lukesjulson/Desktop/github/buzcode/visualization/colormaps/crameri/html
	/Users/lukesjulson/Desktop/github/buzcode/visualization/colormaps/makeColorMap

petersenpeter, 2024-07-18:
Hi Luke

It seems you did not add the CellExplorer subfolders to the Matlab path (you can exclude the folder docs and its subfolders). Go to "Set Path" from the top menu, click the button "Add with subfolders...", and select your CellExplorer folder.

I would also recommend having the CellExplorer folders above your buzcode folders in the set path. 

Let me know how it goes. Happy to chat on zoom as well.

evilrobotxoxo, 2024-07-18:
I see, that's kind of obvious in retrospect. Thanks!

petersenpeter, 2024-07-18:
Happy to help. Let me know if you still experience any issues.

evilrobotxoxo, 2024-07-18:
Thanks, and sorry to bother you with something so silly - it didn’t occur
to me that I needed to set a path for the compiled version. Hope all is
well!

closed 2024-08-13 by petersenpeter