petersenpeter/CellExplorer #114: No NPX 2.0 compatibility?

taningh86, 2024-07-22:
Hi,
Sorry, for posting it also here in issues (I posted in discussions too). Hope that's ok. I have just started to use cellExplorer and trying to figure things out. I am assuming I will have more questions in the future and I hope that's ok. Will very much appreciate your help in this process.
The first question I have for now is if 4-shank neuropixels probe is supported in CellExplorer? I only see IMEC NPX 1.0. But if it is possible somehow then how to add the electrode groups. Is there a guide to it? I am not seeing in the documentations. Please let me know.

Thanks
Jimmy

**Operating system**
Windows 10

**Matlab version**
2023b

**Describe the bug**
A clear and concise description of what the bug is.

**Matlab error log**
Unrecognized field name "electrodeGroups".

Error in ProcessCellMetrics (line 219)
if any([session.extracellular.electrodeGroups.channels{:}]==0)

**Session struct**
If relevant, please provide your session Matlab struct (`basename.session.mat`) by zipping it and dragging it into this text field. This will ease the troubleshooting. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**Matlab setpath**
A common problem is interferences with other toolboxes. If possible, please provide your search path by typing `path` in Matlab.

petersenpeter, 2024-08-13:
Hi 

Please see my reply in the discussion: https://github.com/petersenpeter/CellExplorer/discussions/113

petersenpeter, 2024-08-13:
I am closing this issue, but we can continue in the discussion section.

closed 2024-08-13 by petersenpeter