petersenpeter/CellExplorer #119: Will mountainsort be supported?

ajcheng1, 2024-11-07:
I read in the paper that Mountainsort is supported but under loadSpikes.m it says "mountainsort output format not implemented yet"

ajcheng1, 2024-11-15:
Gotcha. My mountainsort data is unconventionally from preprocessed converted data from numpy, but happy to share. I was able to export it to Phy and then upload it to CellExplorer, but the filtered waveform graph is not showing the individual spikes accurately. Let me know how I can best send you the data (~a couple of gb)

petersenpeter, 2024-11-15:
If you have it in the phy format, you can easily import it into CellExplorer. Could you elaborate on the issue with the filtered waveforms? 

I cannot host the data, so you must find a way to share it.

ajcheng1, 2024-11-15:
Actually nvm about that! I realized I was looking at single waveform and not waveforms (all). 
Slightly related question, given I used mountainsort for spike sorting, my raw data is from numpy. Is it possible for CellExplorer to read .npy files as the raw data instead of .dat?

petersenpeter, 2024-11-28:
Yes, but it doesn't currently support it. I am not sure it is wise to use numpy instead of a binary format, though.

closed 2024-11-28 by petersenpeter