petersenpeter/CellExplorer #124: Now, Kilosort4 didn't output rez.mat, whether influence Neuropixel data into CellExplorer

fdu-lizeyu, 2025-01-27:
**Operating system**
Please provide your operating system

**Matlab version**
Please provide your Matlab version

**Describe the bug**
A clear and concise description of what the bug is.

**Matlab error log**
Please paste your error log here

**Session struct**
If relevant, please provide your session Matlab struct (`basename.session.mat`) by zipping it and dragging it into this text field. This will ease the troubleshooting. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**Matlab setpath**
A common problem is interferences with other toolboxes. If possible, please provide your search path by typing `path` in Matlab.

petersenpeter, 2025-01-28:
Hi 

Indeed, KiloSort4 does not output the rez.mat file, but if you do manual curation in phy, you should not be affected by this. Are you working directly with the output of Kilosort4? 

In CellExplorer, you should define phy as the format, and you should be good.

kkatlowitz, 2025-02-17:
CellExplorer can import spike times but the probe information is loaded in from the rez file.

petersenpeter, 2025-02-17:
Yes, in that case, you can generate it with the session GUI: https://cellexplorer.org/tutorials/channel-maps-tutorial/
Or directly define it in the session struct: https://cellexplorer.org/datastructure/data-structure-and-format/#session-metadata
session.extracellular.chanCoords

kkatlowitz, 2025-02-18:
yup, just defined it directly. apprecaite it.

closed 2025-02-18 by petersenpeter