petersenpeter/CellExplorer #127: [Help] Waveform Plot Problem！

fdu-lizeyu, 2025-04-04:
Waveform shapes in Phy are normal, but in Cellexplorer are abnormal.

[image]

[image]

fdu-lizeyu, 2025-04-05:
For example [image]

petersenpeter, 2025-04-05:
Hi 

I need more info to troubleshoot. But it could be that you did not provide CellExplorer with the correct metadata to extract the spikes. How was the data collected? 

The best option is for you to share a session with me (raw data, spike sorted output from Kilosort(?) and phy, and related metadata).

fdu-lizeyu, 2025-04-06:
Oh Oh, I'm uploading raw data, Kilosort output(version2.5/version4.0) to Google Driver. If completed, I will share URL with you.
Thank you.

fdu-lizeyu, 2025-04-06:
This is website which has raw data and kilosort output(4 and 2.5).
Note: T-shift, highpass filter and CAR have been finished using CatGT before loading in Kilosort. So the data which is loaded in Kilosort has been processed relative to raw data, file 'temp_wh.dat' in folder 'kilosort2_5', file '20250405_mouse01_g0_tcat.imec0.ap.bin' in folder 'kilosort4'. Of course, raw data has been uploaded.
https://drive.google.com/drive/folders/19WAFr_fVtrhGhtaO_lzlpuouumlhyyGb?usp=sharing

fdu-lizeyu, 2025-04-07:
I have already found the reason, and it is related to the selection of the .dat file. I have solved it.
Thank you, Peter. This issue could be closed.

fdu-lizeyu, 2025-04-07:
[image]

closed 2025-04-07 by petersenpeter